helena-project/imix #2: Power LEDs don't come on

alevy, 2016-08-08:
## Cause

This issue is due to a lack of bypass capacitance on the output of the bq24230 (the battery charger and power path IC).
Without a high capacitance on the output, the IC is unstable and does not produce a regulated output voltage.

## Workarounds

The general principle is to add a suitable bypass capacitor to the output of the bq24230 (U1).
There are several ways to achieve this:

1. Plug in the device. With the board face up and oriented like a capital 'T', touch 
the leads of a large (4.7uF to 47uF) capacitor from GND to the upper pad of either R19 or R20.
The LEDS should immediately turn on, and the 3.3V power rail should be working. 
Note that because the capacitor is not permanently attached, the power supply may 
occasionally become unstable again, and the process must be repeated.

2. (recommended) Remove R18 and R19. With the board face up and oriented like a capital 'T', 
solder a large (4.7uF to 47uF) capacitor from the rightmost pad of R18 to the upper pad of R19.
An 0603 capacitor is ideal if possible, since it happens to be the perfect dimensions to bridge the gap.

3. (untested) If no suitable capacitor is on hand, it should be possible to directly apply an input 
voltage that bypasses the power path IC. Solder a wire to the upper pad of R19 (R19 can be removed if desired).
The input voltage at this point should be between 3.4 V and 5.5 V. The chances of this method succeeding are
higher if the bq24230 (U1) is removed, and a 0.1uF to 1uF bypass capacitor is applied on the input rail.

shaneleonard, 2016-08-11:
@alevy Were you able to test any of these workarounds yourself?

alevy, 2016-08-11:
@shaneleonard I have not. I don't have a workbench nearby me, but I'll head into a hacker space sometime soon. I was able to get the LED powered through the JLink connection though -- e.g. when I connect the programmer to and configure it to power the MCU, the LED lights up

shaneleonard, 2016-08-11:
@alevy FWIW, decent soldering irons are $10 at RadioShack, if that's more convenient than going to a hacker space. Otherwise, I could send you one of the boards that I've fixed up. If you were able to power the processor through the JLink, then the only thing preventing you from being able to program the MCU is D8, which shouldn't be too hard to remove with a soldering iron and a reasonably steady hand. When I get the chance, tonight I'll upload my openocd script and a simple blinky app.

shaneleonard, 2016-08-12:
@alevy Uploaded detailed 'getting started' instructions. It turns out that D8 is actually inconsequential. Please take a look at issue #3 and the README for the correct procedure. I've verified this procedure with a totally fresh board, so it should work for you. No need for anything except a JLink and a razor blade.

alevy, 2016-08-12:
Awesome. I'll try it out ASAP

alevy, 2016-08-13:
@shaneleonard I'm able to program the board with your blink app and it works!!

For the record, to get started I need:

  1. Plug USB one of the ports (it doesn't have to be plugged into a power source on the other end_

  2. Provide power from JLink

From that point I can either remove the USB cable and it works fine, or provide power from the USB and stop powering from the JLink. As though the USB + Jlink connectivity is only need to jump-start the process.

alevy, 2016-08-13:
The FTDI chip isn't showing up, but I assume that's related to the short

shaneleonard, 2016-08-13:
@alevy Yay! Did you need to cut the trace? I'm still not exactly sure what the issue is with the FTDI chip. I have one board which I had removed the FTDI from, and the short disappeared, so I have a feeling that there may actually be a different issue with it than I originally thought. How do you configure the JLink to provide power?

alevy, 2016-08-13:
I did have to cut the trace.
I'm not sure how to do it from openocd, but from JLinkExe just use the command `power on`.

closed 2016-11-18 by alevy