helena-project/imix #4: Tag Connect programmer cannot drive the reset line

alevy, 2016-08-08:
## Cause

It is suspected that the solder on the pads of the connector negatively impact the connection enough to
prevent the programmer from sinking enough current to drive the reset line low.

## Workaround

Currently unknown.

alevy, 2016-08-08:
Is it possible to validate this somehow? E.g. by measuring how much current can be drawn through the pad?

FWIW, this seems to _not_ be an issue with the NRF's programming -- at least, I can successfully connect to that chip using my debugger.

alevy, 2016-08-08:
Side question, is this issue dependent somehow on #3? In particular, I haven't yet removed D8 -- will that potentially effect the behavior I see when trying to use the programmer pads?

shaneleonard, 2016-08-11:
@alevy Were you able to connect to the NRF on this revision of Imix?

shaneleonard, 2016-08-11:
@alevy I tried using the same programmer to flash one of my original boards, and encountered the same error. Therefore I suspect that it's actually an issue with my programmer or the toolchain, since the board I'm testing worked earlier this year. I will try using openocd instead of the Atmel toolchain and see what happens. If you are able to resolve issue #2, can you try programming your board and see if it works? You still need to remove D8 because it affects the voltage on the reset line

shaneleonard, 2016-08-11:
@alevy Turns out, the error was Atmel Studio incorrectly configuring the JLink. Using openocd I was able to program the board.

closed 2016-08-11 by shaneleonard

alevy, 2016-08-11:
@shaneleonard I _was_ able to connect to the NRF on this revision of Imix but was not able to connect to the SAM4L - kept getting errors from JLinkExe about not being able to reset the chip. If you were able to though, my guess is that error has to do with the other issues on the board, which I haven't resolved on mine with your workarounds.

Could you post your openocd config files somewhere (this repo) if/when you have a chance? Would be good to switch to it from JLink (since it's open source and easier to install).