helena-project/imix #11: Re-enable RF233 enable pin

alevy, 2016-11-18:
In the current revision, the RF233 enable pin is tied to ground because of an original misdiagnosis of #7. Since that's now fixed we should re-establish the SAM4L's control over the RF233's enable pin.

  - [ ] Test that controlling the enable pin from the SAM4L no longer browns out the SAM4L

  - [ ] If it works, remove the trace/resistor grounding the enable pin.

alevy, 2016-12-22:
In my experiments with an older board --- one without RF233_EN tied to ground, but with the power-related patch wires --- this still does not work. I believe @shaneleonard thinks this is related to the problem in #12 and the same fix might apply.

shaneleonard, 2017-03-12:
Closing as this issue is now tracked by #14

closed 2017-03-12 by shaneleonard