helena-project/imix #13: FTDI pulls reset low by default

alevy, 2016-12-02:
(I'm not sure there is anything great to do about this)

When I attach a terminal to the FTDI, DTR--which is connected directly to the SAM4L's RESET signal--is pulled low, and the board effectively halts. `stormloader` relies on this to be able to reset the board, so this behavior is identical on the Firestorm, but it's a huge bummer because it means that typical utilities for connect to a serial port (e.g. miniterm, minicom, screen, etc) don't work by default.

There are currently two workarounds I've found:

  1. `miniterm2.py` takes a `--dtr` argument which tells it to assert or de-assert the DTR signal. `miniterm2.py --dtr 0 /dev/ttyUSB0 115200` works for me. I'm not sure how to do this with other utilities.

  2. We can program the FTDI chip to invert the DTR signal, which would allow connecting through a terminal serial applications to work out of the box, _but_ the reset signal will be held low by default, so the board won't run.

How does Arduino manage this? Thoughts?

ppannuto, 2016-12-02:
Arudino manages this in hardware. It uses a DTR transition to trigger a
timer, so a reset pulse is emitted on a DTR transition, but the steady
state of DTR is ignored.

shaneleonard, 2017-02-13:
After testing with miniterm, Amit and I concluded that the current behavior is identical to `hail`, and I'm not planning to fix this for the next revision. Leaving it open and labeled as `wontfix` for further potential future research.

alevy, 2017-02-13:
@shaneleonard I agree with this assessment, I think it's fine. I'm closing.

shaneleonard, 2017-02-14:
@ppannuto If I AC-couple the DTR line to the RESET line (as Arduino does), would it be possible to get the behavior we want? It actually seems like a trivial fix (one capacitor), but I don't know if it would cause an undue overhaul in the tockloader or if it would break compatibility with Hail.

shaneleonard, 2017-04-28:
@ppannuto @alevy After some experiments, I found by that removing the reset pullup resistor and replacing the DTR-RESET resistor with a 1uF cap, the DTR can pull down the reset line when desired but the steady-state doesn't make a difference. I was able to successfully get a terminal with all different configurations of `miniterm` and `screen`. `tockloader listen` works correctly, so I'm assuming that the other tockloader functions will work if I use the bootloader. Unless there are any objections, I'm going to make this change.

shaneleonard, 2017-04-28:
I think we should replicate this design on Hail, by the way, since it fixes all of the undesired behavior.

ppannuto, 2017-04-28:
Interesting (and cool, nice work). My understanding is that Arduino adds the unbalanced resistors to lengthen the reset pulse. What does the reset pulse with just the cap look like (duration, especially below reset threshold?). The SAM4L datasheet (sec 42.10.1 RESET_N Timing) specifies a minimum pulse width of 10ns (which upon looking up is wonderfully short, so I suspect the coupling approach works well)

[image]
(I have just no idea what to make of that footnote though; would make me inclined to be comfortably longer than 10ns pulse)

alevy, 2017-04-29:
(this is pretty far out of my comfort zone, so I'm just listening with interest, not commenting)

ppannuto, 2017-05-01:
@shaneleonard could you grab a scope trace probing both sides (DTR and RESET_N) of the cap?

@brghena assuming that looks good, moving hail over to a cap probably a good improvement

shaneleonard, 2017-05-01:
Yeah, ran that experiment last night but forgot to snap a pic. When DTR transitions low, Reset immediately goes to 0V and exponentially rises with a rise time of about 80 ms (it's logic low for about 20 ms). When DTR transitions high (after Reset has resettled high), Reset spikes to 4.2V and decays back to 3.3V. I'll post the screenshot once I take it again.

shaneleonard, 2017-05-01:
The 1uF seems to be overkill, so if it's interesting, I can try a handful of values. I have 470nF and 100nF on hand (although 100nF didn't seem to work when I first tried it)

shaneleonard, 2017-05-01:
@ppannuto @brghena 

[image]

This first picture shows DTR (yellow) transitioning high to low. The reset line (blue) is logic low for 8 ms (shown by the cursors)

shaneleonard, 2017-05-01:
[image]

Here is a zoomed-out version of the same capture.

shaneleonard, 2017-05-01:
[image]

Here, DTR is transitioning low to high. The reset line hits 4.24V at its maximum. Note that DTR doesn't have as sharp of a transition here.

shaneleonard, 2017-05-01:
Sorry for the cell-phone photography by the way. Couldn't find my flash drive to plug into the oscilloscope :/

shaneleonard, 2017-05-01:
@ppannuto @brghena Let me know if you would like me to repeat this for 470nF and/or 100nF.

ppannuto, 2017-05-01:
I think that cap choice is fine. Strictly speaking, it'd probably be good to double check in the datasheet that the reset pin is 4.2V tolerant and that this doesn't wear on the chip (o/w you'd need a clamping diode), but I suspect that's a low enough and short enough transient that it's not a big deal.

@brghena pointed out that when @daniel-scs gets USB support up an running, we can just incorporate that into the bootloader, and drop the FTDI chip altogether, so likely not worth propagating this change to hail. But since you're already spinning a new rev of imix, it looks good to me.

shaneleonard, 2017-05-01:
Datasheet doesn't mention reset being tolerant to a particular level, it just gives a flat 3.6V as the maximum operating voltage for the power domains and I/O. I can add a clamping diode.